LennartHennigs/ESPTelnet #1: Possibility to receive from client?

Teddyz, 2020-07-03:
Great library that allows me to know when someone has connected. 
I am missing printf, but can live with that. 

But it would be great if you could add so we can receive commands from the client. The ctrl-q that is mentioned when connected does not do anything.

Thanks!

LennartHennigs, 2020-07-08:
Hey Teddy,
the idea of this library was to have a way to get status messages from a remote device. The ctrl-q in the example is a reminder (mainly for myself) on how to exit _telnet_. 

I am not sure I understand your request. 
What commands should the client be able to send?
Should the server listen for text input from a client and work with this?

Cheers
l.

Teddyz, 2020-07-10:
Hello Lennart

Yes, the server to listen for commands is what I am looking for. Then I could add: 
```C++
void onTelnetTextLine(String txt) {
    if (txt == "rst") {
        ESP.reset();
    }
    else if (txt == "vars") {
        telnet.printf("vara=%d\r\n", vara);
        telnet.printf("varb=%d\r\n", varb);
        telnet.printf("varx=%d\r\n", varx);
    }
    else {
        printf("Sorry, I did not understand \'%s\'", txt.c_str());
    }
}
```
I was so into this idea that I actually added a few lines (from one of all TelnetStream2 repositories) to achieve this into your code.
```C++
    // check if any incoming data is available to read
  if (client && client.available() && on_data_available != NULL) {
      on_data_available();
    }
    yield();
  } 
/* ------------------------------------------------- */
void ESPTelnet::print(String str) {
  if (client && client.status() == ESTABLISHED) {
    client.print(str); 
  }
}
/* ------------------------------------------------- */
String ESPTelnet::readString() {
  if (client && client.status() == ESTABLISHED && client.available()) {
    return client.readString();
  }
  return String("");
}
bool ESPTelnet::available() {
  if (client.available()) {
    return true;
  }
  return false;
}
```
And it somewhat works. There is a delay of roughly 4 seconds from last character I press in putty.exe until the ESP says it is available. Using Wireshark on my PC reveals that it is immediately sent. So Either my router or the ESP holds the data for a while.
Second problem was that there is some telnet-negotiations from the client (putty) to the server. I had to filter them out too. 

So now I am no longer sure it is so easy to get two-way telnet communication to work smoothly. I have this proof of concept that I can live with. You may close this ticket if you do not find it interesting.

LennartHennigs, 2020-07-10:
Hey Dick,
I like the idea. And thanks for the code. 
Give me a couple of days to test it out myself. Let’s see what I can come up with. 

Cheers.

LennartHennigs, 2021-02-02:
Hey, it took me a while to finally touch this class again but now I added a the option to provide a callback function via `onInputReceived()`. You can see how it works in the example. I experienced no delay.

closed 2021-02-02 by LennartHennigs