LennartHennigs/ESPTelnet #2: Support for ESP32 should be added

mrdc, 2020-07-09:
Hello,

Would be nice to add ESP32 support :)

BTW Does your library can forwards all info from the serial including verbose output?

LennartHennigs, 2021-02-02:
Hey, I finally updated the class. 
ESP32 support should work now. I'll test it tomorrow and then release it if it does.

To your second question I cannot capture Serial.println() output - a good explanation is given [here](https://forum.arduino.cc/index.php?topic=588988.0). But I can use serial input to send to the telnet, see the example, but this is most likely not what you are looking for. You'll have to replace your Serial.print with telnet.print.

LennartHennigs, 2021-02-03:
Tested it and tweaked it a bit. 
Now it runs on a M5Stack without problems.

closed 2021-02-03 by LennartHennigs