LennartHennigs/ESPTelnet #3: Not Issue - Question

svdrummer, 2021-02-04:
Hello :)
As there is a significant change from the last Example on the last version, ie the wifimanager has been removed.
Was there an issue with using wifimanager with this current version?
I am asking as I would like to put the wifimanager back in.
Thank you in advance.

LennartHennigs, 2021-02-04:
Hey, no there isn't (well, almost) if you run it on an ESP8266. WifiManager does not work with ESP32. That's why I removed it in the example. 

But if you stay on the 8266 you're good.

svdrummer, 2021-02-04:
The old example with new library, no longer compiles.
Wemos d1 mini

LennartHennigs, 2021-02-04:
Hey,

you need this on top.

```
#include "WiFiManager.h"

#define PORTAL_TIMEOUT  10 * 60 // seconds
#define AP_NAME         "MY CAPTIVE PORTAL"
#define AP_PASSWORD     ""
```

Then add this function above the `setup()`:

```
void useWiFiManager() {
  // wifiManager.resetSettings();  // this will delete all credentials
  wifiManager.setDebugOutput(false);
  wifiManager.setConfigPortalTimeout(PORTAL_TIMEOUT);
  wifiManager.setAPCallback([] (WiFiManager *myWiFiManager) {
    Serial.println("- No known wifi found");
    Serial.print("- Starting AP: ");
    Serial.println(myWiFiManager->getConfigPortalSSID());
    Serial.println(WiFi.softAPIP());
  });
  // enable autoconnect
  if (!(AP_PASSWORD == "" ? 
    wifiManager.autoConnect(AP_NAME) : 
    wifiManager.autoConnect(AP_NAME, AP_PASSWORD))
   ) {
    Serial.println("- Failed to connect and hit timeout");
    ESP.reset();
    delay(1000); 
  // known wifi found & connected
  }
}
```

And finally replace line 127
`connectToWiFi(WIFI_SSID, WIFI_PASSWORD);`

...with...
`void useWiFiManager() `

That should do the trick.

LennartHennigs, 2021-02-05:
Added the example.

closed 2021-02-05 by LennartHennigs