LennartHennigs/ESPTelnet #4: Build error: #else if in header should be #elif

AFontaine79, 2021-03-24:
Note sure if this works differently on different systems.  I am building in PlatformIO for ESP32 DevKitC with following settings:

[env:esp32dev]
platform = espressif32
build_type = debug
debug_tool = esp-prog
board = esp32dev
framework = arduino
lib_deps = lennarthennigs/ESP Telnet@^1.1.0

I changed
`#else if defined(ARDUINO_ARCH_ESP8266)`
to
`#elif defined(ARDUINO_ARCH_ESP8266)`

LennartHennigs, 2021-03-24:
Ah, I'll thx. I'll change this. (It works on the Arduino IDE.)

closed 2021-03-24 by LennartHennigs

AFontaine79, 2021-04-08:
Hi Lennart, I was able to verify what's happening in the Arduino IDE.  I realized that PlatformIO is showing a warning but building anyway.  This is what the Arduino IDE is doing, except that it normally suppresses warnings.  To see the warning in Arduino, go to File -> Preferences and make sure the "compilation" option is checked for "Show verbose output during:".  Also, make sure compiler warnings is set to "More" or higher.  After you do this and rebuild, you should get the following warning.
`D:\Users\AFont\Documents\Arduino\libraries\ESP_Telnet\src/ESPTelnet.h:19:7: warning: extra tokens at end of #else directive [-Wendif-labels]
 #else if defined(ARDUINO_ARCH_ESP8266)
       ^
`
There is no such thing as `#else if`, only `#elif`.  Everything following `#else` is not evaluated, which is why gcc calls them extra tokens.  See documentation [here](https://gcc.gnu.org/onlinedocs/gcc-3.0.1/cpp_4.html#SEC40).  In this case, it doesn't end up affecting the build.