LennartHennigs/ESPTelnet #5: Build error: WiFiClient::status() not defined on ESP32

AFontaine79, 2021-03-24:
The WiFiClient class does not appear to be cross-compatible between ESP8266 and ESP32.  Annoying, yes, I know.
Anyway... see here:
For the ESP8266, the declaration of status() method is on line 55 of [WiFiClient.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.h).
For the ESP32, [WiFiClient.h](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.h) has no status() method.
I'm just going to modify the code to work around this for now.

LennartHennigs, 2021-03-24:
Hey, I think there is a status() method for the ESP32. Its used in their example:

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClient/WiFiClient.ino

AFontaine79, 2021-04-08:
You are correct.  You actually fixed this issue in commit 830fd75 on Feb 3rd.  My guess is that I somehow grabbed the wrong library version in PlatformIO, which has its own library manager.  Like you said, it builds fine in the Arduino IDE.  I went back and asked PlatformIO to revert my changes to ESPTelnet, and now it looks the same as here.  I am closing this issue.

closed 2021-04-08 by AFontaine79