LennartHennigs/ESPTelnet #6: Feature change: Inherit ESPTelnet from WiFiClient so that it is a Stream

AFontaine79, 2021-03-24:
Right now, ESPTelnet contains WiFiClient as one of its members.  This is a _has a_ relationship.
WiFiClient itself inherits from Stream as follows:
`WiFiClient` --> `ESPLwIPClient` --> `Client` --> `Stream`
If ESPTelnet inherited from WiFiClient, then code like the following would be unnecessary:
```
void ESPTelnet::print(String str) {
  if (client && client.status() == ESTABLISHED) {
    client.print(str); 
  }
}
```
This may be a little tricky to figure out since the WiFiClient object is returned by the WebServer object.  Doing `client = ` is fairly straightforward, but I don't think you can just do `this = ` inside a member function.  I'm going to think about this a bit.

Anyway, the motivation is that having this a Stream object makes it fully compatible with Serial.  I want to be able to pass a generic Stream to a CLI module and not have it be concerned with whether the transport is telnet or serial.

Would you be open to me working on this and potentially contributing some updates to this library?

LennartHennigs, 2021-03-24:
Hey Aaron,
That would be kind of neat. Sure, take a stab at it. 

That's why I have all the stuff online - for people to use and tinker with it... 

Cheers
l.

AFontaine79, 2021-04-07:
Hi Lennart,
Sorry I've been slow to respond on this.  It's been a quite busy couple of weeks.
I would like to take a stab at the changes.  One thing I notice is that WiFiClient already inherits from Stream.  It seems that rather than ESPTelnet inheriting from Stream, it should return a reference to the current connection.  I'm going to put some consideration into this as I work on the CLI module.

A.

SergiuToporjinschi, 2021-07-11:
Nice and clean lib. Returning a client ref it would help a lot to output serial on telnet.

RichK2015, 2021-11-14:
[Nov 21]  Has this work completed?  If so, what version of the Library (will it) appear in.  I's seeing Version 1.2.1 on Arduino

LennartHennigs, 2021-11-15:
no, not completed. there is a working branch that uses it. still need to decide how to split/branch out functionality.

kantat, 2022-09-26:
// send indvidual characters
    } else {
      if (on_input != NULL) {
        if (input.length()) {
          on_input(input + c);//**
          input = "";
        } else {
          on_input(String(c));
        }
      }
//**
on_input(input + String(c));

LennartHennigs, 2022-10-03:
Hey @kantat,
please help me understand your comment.
You changed lines 125ff in [ESPTelnet.c](https://github.com/LennartHennigs/ESPTelnet/blob/master/src/ESPTelnet.cpp) from this...

```
// send individual characters
} else {
      if (on_input != NULL) {
        if (input.length()) {
          on_input(input + c);
          input = "";
        } else {
          on_input(String(c));
        }
      }
```
...to this...

LennartHennigs, 2022-11-20:
There is now an `ESPTelnetStream` class in version 2.0.0 that provides the mentioned functionality.
Thanks again to @AFontaine79 for providing the base for it!

closed 2022-11-20 by LennartHennigs