LennartHennigs/ESPTelnet #9: Passes garbage to on_input if first input string is empty.

alex-fu27, 2021-06-25:
ESP Telnet passes garbage to the on_input callback if the first input string is empty and only `\n` is sent via telnet.

To reproduce, compile and upload:
```c++
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPTelnet.h>

ESPTelnet tnServer;

void setup()
{
	Serial.begin(9600);

	Serial.write("setting up wifi...\n");
	WiFi.hostname("hostname");
	WiFi.begin("wifi", "password");
	WiFi.setAutoReconnect(true);
	while (WiFi.status() != WL_CONNECTED) {
		delay(100);
	}

	Serial.write("setting up telnet...\n");
	tnServer.onConnect([](String){
			tnServer.print("yo ");
			});
	tnServer.onInputReceived([](String s){
			Serial.write("garbage: '");
			Serial.print(s);
			Serial.write("'\n");
			});
	tnServer.begin();

	Serial.write("done\n");
}

void loop()
{
	tnServer.loop();
}
```

Then connect via serial monitor, connect with Telnet and just hit enter on the `yo ` prompt. You will see something like `garbage: '�������� ��!��"��'` in the serial monitor.

LennartHennigs, 2021-07-01:
Hey, thanks for pointing this out. I will take a look...

LennartHennigs, 2021-11-15:
I could only reproduce this on ESP8266, not on ESP32.
Added code, that now manually flushes the client upon connection – and I filter for ASCII chars...

closed 2021-11-15 by LennartHennigs