LennartHennigs/ESPTelnet #11: telnet.onDisconnect don't work

zenbooster, 2021-09-25:
Checked for examples: TelnetServerExample and TelnetServerWithWiFiManager. I connect to the telnet server on esp32 using the putty program in telnet client mode.
When disconnecting from the server, in the case of TelnetServerExample, the onDisconnect handler is never called at all. In the case of TelnetServerWithWiFiManager, the onDisconnect handler is called only the first time, and even then not always (((

zenbooster, 2021-09-26:
I understood what was the matter! =) In the function:
`void ESPTelnetDbg :: loop ()`
there is a line:
`if (client && isConnected &&! isClientConnected (client)) {`

Here you work with the `client` object as if it were a pointer, and you check to see if it is null before calling `isClientConnected`. But in fact, here it is called:
`WiFiClient :: operator bool ()`, which returns true if the socket is not closed. This gives rise to a contradiction: it cannot be that the socket is open and the client is not connected at the same time. I replaced this line with the following:
`if (isConnected &&! (client || isClientConnected (client))) {`
And now everything works! ))

LennartHennigs, 2021-09-26:
Hey,
thanks a lot for finding the bug and suggesting a fix! 
Really appreciate it! 
Tested it real quick and it works like a charm.

Updated the library and pushed the changes.
Thx! 
Have a great weekend.
Cheers
l.

closed 2021-09-26 by LennartHennigs