LennartHennigs/ESPTelnet #12: TelnetServerExample: error: 'class IPAddress' has no member named 'isSet'

zenbooster, 2021-09-26:
When compilling for esp32-wroom-32:

```
C:\Users\home\Documents\Arduino\libraries\ESP_Telnet\src\ESPTelnet.cpp: In member function 'bool ESPTelnet::begin()':
C:\Users\home\Documents\Arduino\libraries\ESP_Telnet\src\ESPTelnet.cpp:16:56: error: 'class IPAddress' has no member named 'isSet'
   if (WiFi.status() == WL_CONNECTED || WiFi.softAPIP().isSet()) {
                                                        ^
```

LennartHennigs, 2021-09-26:
Ah, thanks again. 
Yes, I encountered that before - the WiFi classes of 32 and 8266 are slightly different. That's quite annoying. Plus, it's not that well documented. 

Will check how to determine wether an IP is (in)valid on a 32.

PacciK, 2021-11-10:
Hi @LennartHennigs , I was testing your library on an ESP32 and encountered this exact issue too.
Do you have an update ? Thanks!

RichK2015, 2021-11-14:
This library could be the answer to my prayers.

However, I'm seeing the same isSet() error reported above using my ESP-32 WROOM32D.  I'm using the DOIT DEVKIT V1 board template on Arduino 1.8.51.

Thanks for any help you can provide

NashAli, 2021-11-15:
C:\...\Arduino\libraries\ESP_Telnet\src\ESPTelnet.cpp: In member function 'bool ESPTelnet::begin()':
C:\...\Arduino\libraries\ESP_Telnet\src\ESPTelnet.cpp:16:56: error: 'class IPAddress' has no member named 'isSet'
   if (WiFi.status() == WL_CONNECTED || WiFi.softAPIP().isSet()) {
_________________________________________________________^
I encountered this conflict as well, running an esp32 Dev Module, Win 10 environment using Arduino IDE 1.8.16.

If I compile for the ESP8266 it compiles fine and if I switch to the ESP32, it fails.
Telnet is a very cool feature for the ESP, great work!!
Thanks for your time!!
Nash.

LennartHennigs, 2021-11-15:
Added a private function that either uses `isSet()` (for ESP8266) or looks at the toString() ip address (for ESP32)

closed 2021-11-15 by LennartHennigs

szilvasyz, 2021-12-05:
Dear Friend,
I think you should add the prototype of the _isIPSet() helper function also to the header file. Without this I got the following error message:

/Users/szilva/Documents/Arduino/libraries/ESP_Telnet/src/ESPTelnet.cpp:14:38: error: no 'bool ESPTelnet::_isIPSet(IPAddress)' member function declared in class 'ESPTelnet'
 bool ESPTelnet::_isIPSet(IPAddress ip) {
                                      ^

LennartHennigs, 2021-12-06:
Hey, I published the code with v.1.2.2.
It should work now.