LennartHennigs/ESPTelnet #14: Printf functionality

nicksolark, 2021-11-16:
Could there be a simple way to emulate printf? Not the exact syntax, but the idea.

`DEBUG("variable x value is {0} and y variable is {0}", x, y);`

not sure if it should be {0} and {1}. or is this already supported?

LennartHennigs, 2021-11-17:
Good point. 
After a bit of research I came up this for serial:

```
#define DEBUG_SER_PRINTF(fmt, ...)  \
      do { \
        DEBUG_SERIAL.print(DEBUG_PREFIX); DEBUG_SERIAL.printf(fmt, __VA_ARGS__); \
      } while (0)
  
```

For Telnet I still need to see what to do. 
Might take a while...

DadelNet, 2021-12-05:
I can't wait to be able to use printf with Telnet ... Looking forward to it.

baltazorr, 2022-12-28:
Here is the fix: https://github.com/LennartHennigs/ESPTelnet/pull/35

LennartHennigs, 2023-02-17:
add the the PR #35 to the codebase

closed 2023-02-17 by LennartHennigs