LennartHennigs/ESPTelnet #18: Feature request: change port, disconnect client

modest1981, 2022-02-20:
Hello
Is there any way I can change the default listening port (23)?

Is there any way I can disconnect the client?

LennartHennigs, 2022-02-20:
Hey,
Not yet. 

For now you could change the port in the .h file in line 47. and you could add yourself a function where you call client.stop() as in line 75.  

But valid points, will add them on my todo list on things to update. 

Cheers
l.

modest1981, 2022-03-03:
Thanks for the help but I can't understand the second part. Could you explain more in detail?

LennartHennigs, 2022-03-03:
Hey, just pushed a change.  
You can now define a custom port via the `begin(uint16_t port)` function. 
And added `clientDisconnect()`.  I use them  one in the standard example.

closed 2022-03-03 by LennartHennigs

LennartHennigs, 2022-03-07:
This is now part of 1.3.0

modest1981, 2022-03-09:
Great. Thank you.