LennartHennigs/ESPTelnet #19: _isIPSet(IPAddress ip) problem

bianchina3, 2022-03-24:
Hello.
the line of code:
return ip.toString() == "0.0.0.0";
return always "false".

I have patching with this code:
esp_err_t err;
wifi_mode_t wmode;
err = esp_wifi_get_mode(&wmode);
if(err != ESP_OK) {
   return false;
}
return(wmode == WIFI_MODE_AP);

LennartHennigs, 2022-04-13:
Hey, 
could you please try out the following code:

```
#include <WiFi.h>

IPAddress server_ip;
IPAddress local_ip;

bool isWiFiConnected() {
  return WiFi.status() == WL_CONNECTED;
}

bool _isIPSet(IPAddress& ip) {
  return (ip.toString() != "0.0.0.0");
}

void connectToWifi(const char* ssid, const char* password, int max_tries = 30, int pause = 100) {
  int i = 0;
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (!isWiFiConnected() && i <= max_tries) {
    delay(pause);
    i++;
    Serial.print(".");
  }
  Serial.println("");
  if (isWiFiConnected()) {
    WiFi.setAutoReconnect(true);
    WiFi.persistent(true);
    WiFi.setSleep(false);
  } else {
    Serial.println("Error");
  }
}

void setup() {
  Serial.begin(115200);
  delay(200);
  Serial.println();
  Serial.println("WiFi Test");
}

void loop() {
  if (!isWiFiConnected()) {
    Serial.println("NOT CONNECTED");
  } else {
    Serial.println("CONNECTED ");
  }
  delay(100);
  server_ip = WiFi.softAPIP();
  local_ip = WiFi.localIP();
  
  Serial.println(server_ip.toString());
  Serial.println(_isIPSet(server_ip));
  Serial.println(_isIPSet(local_ip));
  Serial.println();
  delay(1000);
  if (!isWiFiConnected()) {
    connectToWifi("SSID", "PWD");
  }
}
```

The line in question works for me there:

```
WiFi Test
NOT CONNECTED
0.0.0.0
0
0

.......................
CONNECTED 
192.168.4.1
1
1

CONNECTED 
192.168.4.1
1
1
```

LennartHennigs, 2022-04-13:
...but I also realize the correct statement in the lib should be:
```
return ip.toString() != "0.0.0.0"
```

closed 2022-04-15 by LennartHennigs