LennartHennigs/ESPTelnet #20: Add option to use virtual method overrides instead of static callback handlers

sigvind, 2022-05-22:
Would be nice to use class ESPTelnet by inheriting from it and implement a set of virtual methods, instead of having to register static callback methods that don't know anything of the Telnet object.

Add these empty member methods to the ESPTelnet class:
  virtual void onConnectVirtual( String s ) {};
  virtual void onConnectionAttemptVirtual( String s ) {};
  virtual void onReconnectVirtual() {};
  virtual void onDisconnectVirtual() {};
  virtual void onInputReceivedVirtual {};

Then call them where the callback methods are called today.
This would be backwards compatible and improve C++ style usage.

LennartHennigs, 2022-05-26:
Hey, thank you for your comment.
In general, I agree to your point – providing virtual functions and use inheritance would result in "better" C++ code. 
And there are no other benefits besides from having better code, IMO. 

In my experience a lot of Arduino users are not really "fluent" with C++.
Inheritance could be a complex concept to grasp for them. 
This just raises the hurdle of using the class.
And  callbacks don't really need to "know" any internals from the class. 
Plus, this would be a breaking change for the existing user base.

So, I'll keep the class as is for now.
Cheers
l.

closed 2022-05-27 by LennartHennigs

sigvind, 2022-05-27:
It's your library. But you're wrong in that the suggested extension would be a breaking change. I suggested keeping the possibility for registering callbacks, while also having the option for virtuals. It would open for C++ usage, while keeping the C like API too.
But never mind, I may fork it if I don't back out of it. :-)
Thanks.

LennartHennigs, 2022-05-29:
Hey, 
Ah, point taken. Thanks for getting back to me. I misunderstood your comment. Yes, I could keep the functions plus make add virtual ones. Will think about it. 
Cheers
l.