LennartHennigs/ESPTelnet #24: How can I ask password (Hidden Input) to login?

Katheesh, 2022-07-21:
Please help me to find the correct function to ask for hidden input. please reply as soon as possible. Thanks.

Katheesh, 2022-07-21:
[image]

LennartHennigs, 2022-07-21:
Please post the code you used.

Katheesh, 2022-07-21:
```
void setupTelnet() {  
  telnet.onConnect(onTelnetConnect);
  telnet.onConnectionAttempt(onTelnetConnectionAttempt);
  telnet.onReconnect(onTelnetReconnect);
  telnet.onDisconnect(onTelnetDisconnect);
  
  telnet.onInputReceived([](String str) {
    if(!IS_LOGGED_IN){
      if(TELNET_PSWD == str){
        IS_LOGGED_IN = true;
        telnet.print("DasKlug:> ");
        return;
      }
      telnet.print("Invalid Password. Try again...");
      telnet.print("Enter password to access telnet : ");
      return;
    }
    if(str == "" || str == " "){
      telnet.print("DasKlug:> ");
      return;
    }
    toCommand(1, str);
  });

  if (!telnet.begin(TELNET_PORT)) {
    Serial.println("Unable to start telnet...");
  }
}
```

Katheesh, 2022-07-25:
Thanks, man...

closed 2022-07-25 by Katheesh