LennartHennigs/ESPTelnet #25: Telnet support for ESP32 in AP mode

UdayChaduvula, 2022-08-22:
Hi, I am able to implement the telnet for ESP32 in Station mode, but where as I want to implement telnet for ESP32 in AP mode too.

NashAli, 2022-08-22:
bool YourLib::startAPService(const char* ssidAP, const char* appwd) {
  // Setting the ESP as an access point
  // Remove the password parameter, if you want the AP (Access Point) to be open
  WiFi.softAP(ssidAP, appwd, 13, 0, 1);
  //wait for SYSTEM_EVENT_AP_START
  while(!SYSTEM_EVENT_AP_START){delay(100);}
  if (!WiFi.softAPConfig(IPAddress(192, 168, 5, 1), IPAddress(192, 168, 5, 1), IPAddress(255, 255, 255, 0))) {
    return false;
  }
  IPAddress APIP = WiFi.softAPIP();
  return true;
}

NashAli, 2022-08-24:
@UdayChaduvula I have a great example on using the ESP32 follow link below...
https://www.hackster.io/nash-ali/esp32-network-connector-no-hard-code-method-d52f0c

UdayChaduvula, 2023-04-10:
@LennartHennigs I am not able to establish the telnet communication when the ESP-32 is in Acess Point Mode.

volpi3000, 2023-05-09:
Hi @LennartHennigs ,
Thanks for your work! Do you mind adding an example with the telnet server running on an esp running in AP mode. I also can get it to work, although I tried quite a bit.

LennartHennigs, 2023-05-14:
Hey,
my libraries are hobby projects, I did build them mostly for myself. 
As I am currently not working on a telnet project, it might take some time before I provide an AP example. 
I would need to research this, too.

waltje, 2024-05-19:
@LennartHennigs  I will take this on if you haven't gotten around to it.  I do need this for one of my ESP projects, so might as well do it properly and PR it ;-)

LennartHennigs, 2024-05-20:
Hey. Sure feel free to take this on. 
I built this library to solve a problem I had some time ago. But this is out of my scope.