LennartHennigs/ESPTelnet #26: change to "send indvidual characters"

LennartHennigs, 2022-10-03:
// send indvidual characters
    } else {
      if (on_input != NULL) {
        if (input.length()) {
          on_input(input + c);//**
          input = "";
        } else {
          on_input(String(c));
        }
      }
//**
on_input(input + String(c));

_Originally posted by @kantat in https://github.com/LennartHennigs/ESPTelnet/issues/6#issuecomment-1258463723_

LennartHennigs, 2022-10-03:
Hey @kantat,
please help me understand your comment.

You changed lines 125ff in [ESPTelnet.c](https://github.com/LennartHennigs/ESPTelnet/blob/master/src/ESPTelnet.cpp) from this...

```
// send individual characters
} else {
      if (on_input != NULL) {
        if (input.length()) {
          on_input(input + c);
          input = "";
        } else {
          on_input(String(c));
        }
      }
```
...to this...

```
// send individual characters
} else {
      if (on_input != NULL) {
        if (input.length()) {
          on_input(input + String(c));
```

Could you please explain your change a bit more?
Thx and cheers,
l.

kantat, 2022-10-15:
VS2022 + vMicro. ESP8266.
The compiler cursed at the specified line. Corrected for the proposed entry.
Compiler notes are now missing.
Thanks for YOUR work.
I am retired and this is my hobby.

LennartHennigs, 2022-11-03:
Ah, I see. thank you!

closed 2022-11-03 by LennartHennigs