LennartHennigs/ESPTelnet #27: Not working on none WiFi ethernet connections

Mynogs, 2022-10-09:
In the function "ESPTelnet::begin" stands:
`if (WiFi.status() == WL_CONNECTED || _isIPSet(WiFi.softAPIP())) {`

This means that ESPTenet only works if there is a WiFi connection. I use wired ethernet on my board.
I fixed the problem like this:
```
bool ESPTelnet::begin(bool forceConnected, uint16_t port /* = 23 */) {
  ip = "";
  // connected to WiFi or is ESP in AP mode?
  if (forceConnected | WiFi.status() == WL_CONNECTED || _isIPSet(WiFi.softAPIP())) {

```
It works, but it's not pretty...

Greetings 
Mynogs

LennartHennigs, 2022-10-09:
Hey,

Thanks for your comment. 
Does this line of code create problems, as you don't use WiFi?

`server = WiFiServer(port);`

Cheers
l.

Mynogs, 2022-10-09:
"WiFiServer" is a poorly chosen name. It also works with the wired ethernet without any problems. A better name would be "EthernetServer" or something similar.
I check if I can start telnet by
`ETH.localIP() != IPAddress()`
This is perhaps a universal method...
Greetings
Mynogs

Arnold-n, 2022-10-29:
Many thanks @LennartHennigs for your very useful library, and @Mynogs for starting this issue. I have the same problem and would like to support the solution above, perhaps with two minor changes:
-swap arguments for backward compatibility, and use default value `false` for `forceConnected`, and
-use `||` instead of `|` in expression.

```
bool ESPTelnet::begin(uint16_t port /* = 23 */, bool forceConnected /* = false */) {
  ip = "";
  // connected to WiFi or is ESP in AP mode?
  if (forceConnected || WiFi.status() == WL_CONNECTED || _isIPSet(WiFi.softAPIP()))
```

To me `ethConnected` seems a more descriptive name than `forceConnected`.

LennartHennigs, 2022-10-30:
Hey @Arnold-n,
If find your and @Mynogs request a valid one.
As you seem to be using an ethernet connection, too, could you please verify that this line of code:

```server = WiFiServer(port);```

...in the `begin()` function does not create a problem?
Thx.

Cheers
l.

Arnold-n, 2022-10-30:
Thanks @LennartHennigs - and indeed `WiFiServer` does not create a problem.

I agree with @Mynogs  that the name `WiFiServer` is confusing, I think it should have been called `NetworkServer` as (in my limited understanding of things) it just seems to use whichever network connection is available.

In my code I use

```
#include <WiFiClient.h> // WiFiClient (-> TCPClient), WiFiServer (->TCPServer)
 using TCPClient = WiFiClient;
 using TCPServer = WiFiServer;
```

and setup() uses ethernet if available and if a connection can be made, or else it falls back to WiFi. In both cases telnet is working fine, even after a dis- and reconnect.

```
ESPTelnet telnet;
#include "W5500lwIP.h"
Wiznet5500lwIP eth(CSPIN);

setup() {
  .. /* set up SPI and W5500 ethernet here */
  if (eth.connected()) {
    telnet.begin(23, true);
  } else {
    .. /* set up WiFi here */ ..
    telnet.begin(23);
  }
```

What I didn't achieve yet is a change from ethernet to WiFi if ethernet gets disconnected after setup() finishes (and back when it is reconnected).

For anyone using the code above for a W5500, it is important to apply [this patch](https://github.com/esp8266/Arduino/issues/8498) to the BSP if the W5500 may be absent or unpowered, to avoid a WDT crash upon boot.

LennartHennigs, 2022-11-03:
Hey @Arnold-n and @Mynogs,
I added a second parameter to the `begin()` function:

`bool begin(uint16_t port = 23, bool checkConnection = true);`

This allows you to bypass the WiFi check.

closed 2022-11-03 by LennartHennigs