LennartHennigs/ESPTelnet #32: Telnet username / password login

jamesarm97, 2022-12-07:
Has anyone implemented a username / password login using this library? Would it just be done by parsing the input and not allowing any commands to be issued?

milansim, 2022-12-12:
I've just made it for myself, in case you still need it. It could get more complex but since it is exposed to my Wifi only this is fine for most of us

const char* password = "yourpass" //I use the same for OTA, WifiAP...
boolean passOK = false;
int passretry = 10;
String bannedIP ;

telnet.onInputReceived([](String str) {  // checks for a certain command   !!!!!!!!!!!
    if (passOK == false){
        if (str == password) { passOK = true; info();} else {
          passretry--; 
          if (passretry==0){
               bannedIP = telnet.getIP(); telnet.println("Fail, IP address banned !"); telnet.disconnectClient(); 
               passretry=10;
          } else {telnet.println("Incorrect ! ,"+ String(passretry) + " more attempts");}
        }
    } else {
        if (str == "info") { info(); } 
        else if (str == "update_firmware") { telnet.println("Updating system, hold on"); OTA_Updates(); }
        ......
          else if (str == "espreset") { telnet.println("> resetting ESP..."); delay(500); ESP.restart();}
    };

void onTelnetConnect(String ip) {
  if (telnet.getIP() == bannedIP) {telnet.disconnectClient();}
  telnet.println("\nWelcome " + telnet.getIP()); telnet.println("Please provide password :");
  passOK = false;
}

LennartHennigs, 2022-12-17:
Hey @milansim,
thank you for sharing this!
Cheers
l.

milansim, 2022-12-19:
this is not really done... every telnet.print() should be replaced with if (passOK == true){ telnet.print();} which would be a large task
Much better would be to implement it in your library !

LennartHennigs, 2024-03-26:
I understand, but implementing logic for what you do want to do with the telnet connection is not in scope for me.

closed 2024-03-26 by LennartHennigs