LennartHennigs/ESPTelnet #34: Is it possible to allow more than one user

svdrummer, 2022-12-17:
Hello, Is it possible to stream this data out to several clients?
A ONE to many situation

Currently I am sending serial port data from a soft_AP server to 10 clients. Same data at the same time.
I am wondering if this can do the, even though the docs do say "Only a single client"
The server is an ESP32

LennartHennigs, 2022-12-17:
Hey, 
generally yes. 
There is a limit to 4 connections for a TCPServer. So 10 might not work.
For now I check in my code the connected IP address as I wanted to make sure that only one client connects.

svdrummer, 2022-12-17:
No problem. Thanks anyway. 10 clients for ESP32 and four for ESP8266. I will find another way

LennartHennigs, 2022-12-17:
I could add a bool to optionally disable the one IP check. I'll take a look soon.

svdrummer, 2022-12-17:
In my situation, I have remote devices just logging in to access a streaming serial port data. Ie, serial data is being fed into the server. When remotes log in, they get access to the serial data.

closed 2023-01-15 by svdrummer