LennartHennigs/ESPTelnet #36: Reconnect problem

ams-hh, 2023-02-07:
Hello !
And at first: Thank you very much for this library!!

I want to establish a message service for debugging purposes between my development system (ESP32 based) and another ESP (ESPxx based). The development system runs your Telnet-Server.

The other ESP connects as a Telnet Client to this server.

Unlike in a program like Putty, which in all cases (eg killing the process via task manager) sends a disconnect, I cannot do that with a stand-alone ESP Telnet client. When I trigger RESET there, it's over and then next thing after rebooting is again the connect attempt.

And there is the problem.
In ESPTelnetBase you programmed

``
void ESPTelnetBase::loop() {
  //check if there are any new clients
  if (server.hasClient()) {
    isConnected = true;
    // already a connection?
    if (client && client.connected() && isClientConnected(client)) {
      TCPClient newClient = server.available();
      attemptIp  = newClient.remoteIP().toString();
      // reconnected?
      if (attemptIp == ip) {
        if (on_reconnect != NULL) on_reconnect(ip);
        disconnectClient();
        client = newClient;
        emptyClientStream();
      // disconnect the second connection
      } else {
        if (on_connection_attempt != NULL) on_connection_attempt(ip);
        return;
      }

``

So if there is a connection attempt from the same IP it calls at first the on_reconnect cb and then
disconnects the client. And after that it does not work any more.

I can see the //disconnect the second connection - comment.
But .... 
If I comment the line "disconnectClient()" everything works as expected.
After rebooting the server ESP:

Starting the client ESP for the first time I get on the server side
- connected
and can receive messages correctly from the client

After resetting the client ESP I get on the server side
- reconnected
and can receive messages correctly from the client (on and on)

If I keep the line "disconnectClient" I get after the client-reset on the server side
- reconnected
- disconnected
and cannot receive any message any more.

What do I wrong? I there a workaround?

Best regards from Hamburg and thank you
Andree

LennartHennigs, 2023-02-26:
Moin Andree,
Grüße nach Hamburg.

By looking at your code snippet I can see that you are not using the latest version of ESPTelnet.
Could you check out the current sources here at GitHub and take them for a spin?
The part you talk about now looks like this:

```
      // yes, reconnected
      if (attemptIp == ip) {
        disconnectClient(false);
        connectClient(false);
        if (on_reconnect != NULL) on_reconnect(attemptIp);
        // no, throw error
      } else {
        if (on_connection_attempt != NULL) on_connection_attempt(attemptIp);
        return;
      }
    }
```

Could you check, that this also does not work for you?

In the meantime, I'll see if there is an elegant way to detect that the client has "left". Quitting out of my telnet client usually does not trigger an event. But let's see...

Cheers
l.

ams-hh, 2023-02-26:
Moinsen Henning!

"...that you are not using the latest version..."
I think it was three weeks ago...?

In the meantime I established another solution by using pure tcp (as you do inside).
Thank you for the response.

Best regards
Andree

LennartHennigs, 2023-02-26:
>  I think it was three weeks ago...?

Yes, it was. 
But as this is a hobby, I address things when I find the time and headspace to do so.

> In the meantime I established another solution by using pure tcp (as you do inside). 
Nice.

If you would still want to give me feedback, I found a way to detect whether a client "silently" disconnects.
You can find it in this branch:

https://github.com/LennartHennigs/ESPTelnet/tree/test-if-alive

I tihnk, this solves your problem.

Cheers
Lennart

ams-hh, 2023-02-26:
"when I find the time and headspace to do so"

Nothing to complain about!!
As you say, it's for hobby purposes!

"I found a way to detect whether a client "silently" disconnects"

Do you mean this piece of code?
`
// frequently check if client is still alive
  if (isConnected) {
    long now = millis();
    if (now - last_status_check >= timeout_interval) {
      last_status_check = now;
      if (client.status() == 0) disconnectClient();
    }
  }

`

LennartHennigs, 2023-02-26:
Yes.

LennartHennigs, 2023-02-27:
Note to myself: works on ESP8266, but the ESP32 has no `client.status()`.

closed 2023-03-12 by LennartHennigs