LennartHennigs/ESPTelnet #41: Callbacks doesn't works correctly with PlaformIO but works in Arduino IDE

AdriRayns, 2023-02-21:
Hello,

I am encountering some strange behavior with the library with Visual Code + PlatformIO, and I am not being able to determine the cause due to lack of experience and time.

Trying to use ESPTelnet in a project, I found a strange behavior with the Callback functions. After many changes and trying to find the reason, I was able to detect the same problem with the examples provided in the ESPTelnet library.
This happens only when the code is compiled and uploaded from Visual Code (PlatformIO). While, for example, "TelnetServerExample" with Arduino IDE works correctly.

I don't know if the cause is a difference of library versions / espressif-framework, or if it is a compile method problem.

### Error description:
The callbacks onInputReceived, onConnected, onDisconnected, etc. do not work correctly. Only the first callback to onConnect() works correctly. After closing the connection (telnet> quit) from the client, onDisconnected is not executed. When creating a new connection, onDisconnect() is executed, followed by onConnect().
### Reproduce the error:
* Copy the "TelnetServerExample" example into a Visual Code + PlatformIO project (ex: main.cpp).
* Connect a Telnet client
* Disconnect the Telnet client
** The callback onDisconnect() is not executed. It is only executed if the client reconnects.
### Platform.ini file:
```
[env:nodemcu]
platform = espressif8266
board = d1
framework = arduino
lib_deps =
	https://github.com/LennartHennigs/ESPTelnet
```
### Dependency graph in build (PlatformIO):
```
framework-arduinoespressif8266 @ 3.30101.0 (3.1.1)
tool-esptool @ 1.413.0 (4.13)
tool-esptoolpy @ 1.30000.201119 (3.0.0)
toolchain-xtensa @ 2.100300.220621 (10.3.0)
---
ESP Telnet @ 2.0.0.0+sha.b7ae843
|-- ESP8266WebServer @ 1.0
|    |-- ESP8266WiFi @ 1.0
|-- ESP8266WiFi @ 1.0
```
Could you shed some light on this matter for me? If I need to provide more data, please let me know.

Thank you!

DadelNet, 2023-02-22:
Hi, 3 points :

1. I have just retested onInputReceived on a Wemos D1 Mini on PIO : user inputs from PuTTY such as a password or commands work with lib_deps = lennarthennigs/ESP Telnet @ ^2.0.0  but not with the latest version (ESP Telnet @ 2.0.0+sha.b7ae843) loaded with : https://github.com/LennartHennigs/ESPTelnet. I can still connect 

2. Also, nothing to do with this revision issue as the same thing happen on 2.0.0 without noticeable effects as far as I can see, but the compilation throws the following error messages :

.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnet.cpp: In member function 'size_t ESPTelnet::printf(const char*, ...)':
.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnet.cpp:152:13: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
  152 |     if (len >= sizeof(loc_buf)) {
Compiling .pio\build\d1_mini\lib7b3\ESP Telnet\EscapeCodes.cpp.o
.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp: In member function 'void ESPTelnetBase::loop()':
.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp:41:46: warning: 'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]
   41 |       TCPClient newClient = server.available();
      |                                              ^
In file included from C:\Users\dadel\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:40,
                 from .pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.h:20,
                 from .pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp:3:
C:\Users\dadel\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiServer.h:85:14: note: declared here
Compiling .pio\build\d1_mini\lib55c\ESP8266mDNS\ESP8266mDNS.cpp.o
   85 |   WiFiClient available(uint8_t* status = NULL) __attribute__((deprecated("Renamed to accept().")));
      |              ^~~~~~~~~
.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp: In member function 'void ESPTelnetBase::connectClient(bool)':
.pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp:66:29: warning: 'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]
   66 |   client = server.available();
      |                             ^
Archiving .pio\build\d1_mini\lib7b3\libESP Telnet.a
In file included from C:\Users\dadel\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:40,
                 from .pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.h:20,
                 from .pio\libdeps\d1_mini\ESP Telnet\src\ESPTelnetBase.cpp:3:
Compiling .pio\build\d1_mini\lib55c\ESP8266mDNS\LEAmDNS.cpp.o
C:\Users\dadel\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiServer.h:85:14: note: declared here

3. Finally, the code seems to crash randomly with the latest version (but not the old one) :

2023-02-22 09:17:23 --> - Telnet
2023-02-22 09:17:23 --> : running
2023-02-22 09:17:52 --> onTelnetConnect : asking password ...
2023-02-22 09:31:22 --> 
2023-02-22 09:31:22 --> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
2023-02-22 09:31:22 --> 
2023-02-22 09:31:22 --> Soft WDT reset
2023-02-22 09:31:22 --> 
2023-02-22 09:31:22 --> >>>stack>>>
2023-02-22 09:31:22 --> 
2023-02-22 09:31:22 --> ctx: sys
2023-02-22 09:31:22 --> sp: 3fffeb70 end: 3fffffb0 offset: 01a0
2023-02-22 09:31:22 --> 3fffed10:  00000000 0fea48c5 40214150 00000913  
2023-02-22 09:31:22 --> 3fffed20:  3fff0d28 000d09ca 00000913 4021415d  
2023-02-22 09:31:22 --> 3fffed30:  000d09ca 00001388 000cff55 402141d3

AdriRayns, 2023-02-22:
Hello,

Thank you guys. I was just coming to tell you that the main reason for the failure between Arduino IDE and PlatformIO is that it was not really the same version of the library as @DadelNet says, setting correctly the library to 2.0.0 works correctly.

LennartHennigs, 2023-02-22:
Hey guyw, 
please check the current version.  I actually found a bug in the `loop()`. 
Text input was not detected after the first letter.
I fixed it and it should now work as expected.

AdriRayns, 2023-02-22:
Hi again @LennartHennigs 
I have just confirmed that current master branch is working as expected

DadelNet, 2023-02-22:
Hi @AdriRayns,
It is late and I must be missing something ! I have "cleaned" the library on PIO and recompiled using  "lib_deps = https://github.com/LennartHennigs/ESPTelnet" but it still defaults to "ESP Telnet @ 2.0.0+sha.b7ae843". How do I get PIO to load the latest version ?

AdriRayns, 2023-02-23:
Hi @DadelNet 

Well, right now being late for me too, I don't know what to tell you. I use the same method, "Clean all" and "build" again.
Make sure you don't already have ESPTelnet (for testing) in ./lib/.
As you rightly say, it is enough with: 

lib_deps=
https://github.com/LennartHennigs/ESPTelnet

I am getting the latest version of master
```
|-- Logger
|   |-- ESP Telnet @ 2.0.0+sha.1764033
|   |   |-- ESP8266WebServer @ 1.0
|   |   |   |-- ESP8266WiFi @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
```

DadelNet, 2023-02-23:
@AdriRayns. Thanks for the tip.

@LennartHennigs. All tested and working. Thanks.

closed 2023-02-23 by LennartHennigs