LennartHennigs/ESPTelnet #43: Obsolete: WiFiServer::available()

hsaturn, 2023-03-11:
Hello

::available is obsolete. Just have to rename it to accept to fix the compilation warnings

I'm using ## [2.0.0] - 2022-11-21 (Changelog)

ESPTelnetBase.cpp:34
ESPTelnetBase.cpp:49

Best regards

LennartHennigs, 2023-03-12:
Hey, 
thank you for reaching out. 
I just treleased 2.1.0 which fixes this waning and adds quite a few enhancements.
Cheers
l.

closed 2023-03-12 by LennartHennigs

hsaturn, 2023-03-12:
By the way, I think that ESPTelnetStream may fit perfectly with my TinyConsole, I'm gonna try this on my "roomba 620" project.
I'm controlling yet the roomba with ESPTelnet that works fine.
Using TinyConsole.begin(ESPTelnetStream) would allow me to add colors and cursors move to my actual telnet session.

Sounds great.