LennartHennigs/ESPTelnet #45: ESPTelnetStream private of Stream

hsaturn, 2023-03-12:
Hello

ESPTelnetStream inherits privately from Stream.

I don't find any reason for this (and this prevent me to do things :-)

Is it possible to replace the inheritance with:

``` c++
 class ESPTelnetStream : public ESPTelnetBase, public Stream 
                                               ^^^^^^
```

(note: tried on clone master from github since the 2.1.0 is not yeat deployed)

Best regards

LennartHennigs, 2023-03-12:
Hey, @hsaturn,
created a branch for this. 
Does it work for you?

hsaturn, 2023-03-13:
Works perfectly on this branch:

$ git status
On branch 45-esptelnetstream-private-of-stream
Your branch is up to date with 'origin/45-esptelnetstream-private-of-stream'.

nothing to commit, working tree clean

[image]

LennartHennigs, 2023-03-13:
ok, cool @hsaturn.
happy to hear. 
merged the branch now and will release it 2.1.1