LennartHennigs/ESPTelnet #47: Additional datatypes overloads ??

billbrach, 2023-06-25:
Not a bug, but something that would reduce the amount of typing for a .print or .println, would be to add a couple of more overloads of the .print and .println function calls for longs like uint32_t

In particular, if I need to debug anything using long datatypes, such as the Arduino millis() statement, I have to type:
telnet.print((String)millis())
instead of:
    telnet.print(millis())

LennartHennigs, 2023-06-26:
Hey Bill,
what version of the library are you using? 
And on what device?

I do not need to typecast on an ESP32 with the current version.
Cheers
l.

billbrach, 2023-06-26:
Thanks for the prompt reply !!  ESP8266, specifically D1 Mini.  Library version is 2.1.1 on Arduino IDE 1.8.13 running Linux Mint 19.  Using telnet to debug RS232 lithium battery controllers, installed in a 60 year old bus, sitting in my driveway !!

LennartHennigs, 2024-03-26:
telnet.print() is now a template function.

https://github.com/LennartHennigs/ESPTelnet/blob/e8bfb78e3c9a5dad064fe7147736d89ba28278de/src/ESPTelnet.h#L19

This issue should no longer occur.

closed 2024-03-26 by LennartHennigs