LennartHennigs/ESPTelnet #50: Can't Get example with AutoConnect to Work

DanMan32, 2023-08-28:
I tried the example TelnetServerWithAutoconnect and I can't get it to do what I believe it is supposed to.
I can get connected to the WiFi alright through the initial Autoconnect AP and from there set up AutoConnect to connec to my WiFi,
And after that I can connect to the Telnet server using Putty.
But all it seems to do is echo what I type,  If I understand the code correctly, I should get informed from the serial connection when I make the telnet connection, what I type in serial should be output to the telnet connection, and if I enter 'ping' in the telnet, I should get a response 'pong', an get disconnected if I enter 'bye' the telnet session should disconnect.
But nope, only echoed characters.

Speaking of echoed characters, I recommend a function to switch echo on and off.   That may satisfy the other request to hide characters typed when entering a password.   There may be other circumstances where one might not want an auto-echo, but be able to control what gets echoed, especially if they are processing input a byte at a time to determine what key was pressed and produce an action according to what was pressed.
Example: esc to go back a menu

DanMan32, 2023-08-28:
P.S. 
Seems to be a flaw in the loop (main) section of the mentioned sketch where when there's a byte in the serial buffer, to echo it back to the terminal.
serial.read() returns an integer, not a character.

LennartHennigs, 2024-03-26:
Hey, 
echoing characters has something to do with the Telnet client negotiation and not with the ESPTelnet.
Please see #64. You must either change the setting in your Telnet client or negotiate the local echo with the client.

LennartHennigs, 2024-03-26:
> Seems to be a flaw in the loop (main) section of the mentioned sketch where when there's a byte in the serial buffer, to echo it back to the terminal.
> serial.read() returns an integer, not a character.

No, `telnet.print()` is a template function, the type is detected and kept intact. 
https://github.com/LennartHennigs/ESPTelnet/blob/e8bfb78e3c9a5dad064fe7147736d89ba28278de/src/ESPTelnet.h#L19

closed 2024-03-26 by LennartHennigs