LennartHennigs/ESPTelnet #58: PlatformIO - error: #if with no expression  #if DEBUG_ON

dewenni, 2024-02-13:
I tried to use this library in my ESP32 project that is programmed with Visual Studio Code and PlatformIO

When I compile the project I get the following compiler error:

```
.pio/libdeps/esp32/ESP Telnet/src/DebugMacros.h:29:13: error: #if with no expression
 #if DEBUG_ON
             ^
```

my platformio.ini
```
[env:esp32]
platform = espressif32@5.4.0
board = upesy_wroom
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder, colorize
build_type = release
board_build.partitions = min_spiffs.csv
```

LennartHennigs, 2024-02-17:
hey, @dewenni,

can you check if this branch still throws an error for you:
https://github.com/LennartHennigs/ESPTelnet/tree/58-platformio-error-if-with-no-expression-if-debug_on

dewenni, 2024-02-17:
Thanks, I will test it asap

dewenni, 2024-02-17:
@LennartHennigs 
I have just tested it and it now compiles without errors. Seems to be solved with the changes.
Thank you very much

LennartHennigs, 2024-02-17:
Ok. Will merge it then.

closed 2024-02-17 by LennartHennigs