LennartHennigs/ESPTelnet #61: Unable to stop

F1p, 2024-02-19:
`void startTelnet(){
  DEBUG_PRINT("Telnet: ");
  if (TelnetServer.begin()) {
    TelnetRunning = true;
    DEBUG_PRINTLN("Running");
  } else {
    DEBUG_PRINTLN("error.");
    //errorMsg("Will reboot...");
  }
}

void stopTelnet(){
  DEBUG_PRINTLN("Stopping Telnet");
  TelnetServer.stop();
  TelnetRunning = false;
}`

Requesting TelnetServer.stop(); doesn't close an active connection, only prevents new?

LennartHennigs, 2024-03-26:
No, the `stop()` function calls `WiFiServer.stop()` which closes the connection:
https://github.com/arduino/esp8266/blob/1eb0645dcb82e03e239e7932ad20802c769c89e0/libraries/ESP8266WiFi/src/WiFiServer.cpp#L132

F1p, 2024-03-26:
I observed different behaviour,

Is it possible to enable/disable the telnet server?

LennartHennigs, 2024-03-26:
This is what `stop()`does.
You can disconnect a client via `disconnectClient(bool triggerEvent = true)` as stated in the documentation

F1p, 2024-03-26:
When calling stop() I found new connections prevented, but already established connections would remain active until terminated by the user