LennartHennigs/ESPTelnet #63: Wired LAN8720 connected check / workaround / fix?

HotNoob, 2024-03-18:
using Eth.H / EthClass.H

which piggy backs a bit off of the wifi events a bit...

this check, which is enabled by default will stop begin from working. 

https://github.com/LennartHennigs/ESPTelnet/blob/e8bfb78e3c9a5dad064fe7147736d89ba28278de/src/ESPTelnetBase.cpp#L17

#include <ETH.h>

``` if ( !ETH.linkUp() &&  WiFi.status() != WL_CONNECTED && !_isIPSet(WiFi.softAPIP())) return false; ```

fixes this problem, alternatively as a work around, turning the check off via bool works too.
 
``` telnet.begin(23, false); ```

it does require making the lib more bulky...  so could maybe use this WITHOUT <ETH.h>. only compatible #if ESP_IDF_VERSION_MAJOR > 3
``` if ( !(WiFiGenericClass::getStatusBits() & ETH_CONNECTED_BIT) &&  WiFi.status() != WL_CONNECTED && !_isIPSet(WiFi.softAPIP())) return false; ```

 instead, to avoid having to import Eth.H for wifi only builds.
  
 works fine over eth \w main for me with some small modification.

LennartHennigs, 2024-03-26:
Hey,
As you pointed out, my answer would have been to disable the check via:

`telnet.begin(23, false);` :-)

By looking at this `#if ESP_IDF_VERSION_MAJOR > 3`, I guess you use an ESP32.

Can you point me to some additional examples/explanations for this part of your suggestion:

```
#if ESP_IDF_VERSION_MAJOR > 3

 if ( !(WiFiGenericClass::getStatusBits() & ETH_CONNECTED_BIT))
 ```

That would allow me to judge better whether I want to include this – as there was a request about some Ethernet board before: #48.

HotNoob, 2024-03-26:
hmm. welp, i'm new to arduino stuff... and rusty af \w c / c++ :P

but, 

i'm just stealing that code from:
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/Ethernet/src/ETH.cpp#L957C12-L957C45

a bit of a hack
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/Ethernet/src/ETH.h#L66

since wifi.h is already including wifigenericclass yaddi yaddie. 
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/WiFi/src/WiFi.h#L34

now that i think about it a bit more. 
could probably be reduced to just:
`WiFiGenericClass::getStatusBits()`

because we don't care if its eth or wifi, just that it's connected. 

---
so.... 

` if ( !WiFiGenericClass::getStatusBits() &&  WiFi.status() != WL_CONNECTED && !_isIPSet(WiFi.softAPIP())) return false; `

should be sufficient. i think #ESP_IDF_VERSION_MAJOR  is probably not necessairy as it's not in expressif's version. 
that part of the code comes from a custom class that came from lilygo which i see is not necessary and is based on Eth.H from expressif. working 100% fine using Eth.H from expressif.

been running the modified code for a bit now, working fine, but donno if it effects functionality if using wifi. 

---
as far as the other issue, #48 , he is using a W5100 chip, which goes over SPI afaik, which i think might use different libraries. 
Ethernet.h is not used for me. 

the LAN8720 im using is built into the board... donno if expressif's Eth.H works for his W5100.

HotNoob, 2024-03-26:
hmm. welp, i'm new to arduino stuff... and rusty af \w c / c++ :P

but, 

i'm just stealing that code from:
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/Ethernet/src/ETH.cpp#L957C12-L957C45

a bit of a hack
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/Ethernet/src/ETH.h#L66

since wifi.h is already including wifigenericclass yaddi yaddie. 
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/WiFi/src/WiFi.h#L34

Wifi.H does include ETH_CONNECTED_BIT; this is where the Eth connected status is set. 
https://github.com/espressif/arduino-esp32/blob/e92b4ca62b7f1375b762ee18ceb9f59dbd4a04dc/libraries/WiFi/src/WiFiGeneric.cpp#L1159
---
so.... 

` if ( !(WiFiGenericClass::getStatusBits() & ETH_CONNECTED_BIT ) &&  WiFi.status() != WL_CONNECTED && !_isIPSet(WiFi.softAPIP())) return false; `

should be sufficient. i think #ESP_IDF_VERSION_MAJOR  is probably not necessairy, as its not on the git repo, but is in the lib from platform io.

been running the modified code for a bit now, working fine. i don't think it'll effect wifi functionality because the bit status is done in the wifi libs. 

---
as far as the other issue, #48 , he is using a W5100 chip, which goes over SPI afaik, which i think might use different libraries. 
Ethernet.h is not used for me. 

the LAN8720 im using is built into the board... donno if expressif's Eth.H works for his W5100.

closed 2026-02-26 by LennartHennigs