LennartHennigs/ESPTelnet #67: v2.2.1 ESP32 with arduino-esp32 v3.0.1

sirpete83, 2024-06-06:
Hi, ESP32_Telnet is not compiling unter v3.0.1

```
In file included from c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h:9,
                 from C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:4:
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:32:19: error: 'WiFiClient' does not name a type; did you mean 'Client'?
   32 | using TCPClient = WiFiClient;
      |                   ^~~~~~~~~~
      |                   Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:33:19: error: 'WiFiServer' does not name a type; did you mean 'Server'?
   33 | using TCPServer = WiFiServer;
      |                   ^~~~~~~~~~
      |                   Server
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:59:3: error: 'TCPClient' does not name a type; did you mean 'Client'?
   59 |   TCPClient& getClient();
      |   ^~~~~~~~~
      |   Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:68:3: error: 'TCPServer' does not name a type; did you mean 'Server'?
   68 |   TCPServer server = TCPServer(23);  // must be initalized here
      |   ^~~~~~~~~
      |   Server
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:69:3: error: 'TCPClient' does not name a type; did you mean 'Client'?
   69 |   TCPClient client;
      |   ^~~~~~~~~
      |   Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:94:22: error: 'TCPClient' has not been declared
   94 |   void connectClient(TCPClient c, bool triggerEvent = true);
      |                      ^~~~~~~~~
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:99:27: error: 'TCPClient' has not been declared
   99 |   void handleReconnection(TCPClient &newClient, const String &attemptedIp);
      |                           ^~~~~~~~~
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnetBase.h:100:33: error: 'TCPClient' has not been declared
  100 |   void handleExistingConnection(TCPClient &newClient);
      |                                 ^~~~~~~~~
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h: In member function 'void ESPTelnet::print(const T&)':
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h:20:9: error: 'client' was not declared in this scope; did you mean 'Client'?
   20 |     if (client && isConnected()) {
      |         ^~~~~~
      |         Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h: In member function 'void ESPTelnet::println(const T&)':
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h:31:9: error: 'client' was not declared in this scope; did you mean 'Client'?
   31 |     if (client && isConnected()) {
      |         ^~~~~~
      |         Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h: In member function 'void ESPTelnet::print(const T&, int)':
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h:42:9: error: 'client' was not declared in this scope; did you mean 'Client'?
   42 |     if (client && isConnected()) {
      |         ^~~~~~
      |         Client
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h: In member function 'void ESPTelnet::println(const T&, int)':
c:\_PPM\Arduino_WS\libraries\ESP_Telnet\src/ESPTelnet.h:53:9: error: 'client' was not declared in this scope; did you mean 'Client'?
   53 |     if (client && isConnected()) {
      |         ^~~~~~
      |         Client
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino: In function 'bool isConnected()':
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:36:11: error: 'WiFi' was not declared in this scope
   36 |   return (WiFi.status() == WL_CONNECTED);
      |           ^~~~
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:36:28: error: 'WL_CONNECTED' was not declared in this scope
   36 |   return (WiFi.status() == WL_CONNECTED);
      |                            ^~~~~~~~~~~~
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino: In function 'bool connectToWiFi(const char*, const char*, int, int)':
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:43:3: error: 'WiFi' was not declared in this scope
   43 |   WiFi.mode(WIFI_STA);
      |   ^~~~
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:43:13: error: 'WIFI_STA' was not declared in this scope; did you mean 'WIFI_SSID'?
   43 |   WiFi.mode(WIFI_STA);
      |             ^~~~~~~~
      |             WIFI_SSID
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino: In function 'void setup()':
C:\WINTEMP\.arduinoIDE-unsaved202456-21580-7l55av.231sx\AnsiExample\AnsiExample.ino:149:10: error: 'WiFi' was not declared in this scope
  149 |     ip = WiFi.localIP();
      |          ^~~~
```

Bibliothek WiFi in Version 2.0.0 im Ordner: C:\_PPM\Arduino_WS\hardware\espressif\esp32_v3\libraries\WiFi  wird verwendet
Bibliothek Networking in Version 1.0.0 im Ordner: C:\_PPM\Arduino_WS\hardware\espressif\esp32_v3\libraries\Network  wird verwendet
Bibliothek ESP Telnet in Version 2.2.1 im Ordner: C:\_PPM\Arduino_WS\libraries\ESP_Telnet  wird verwendet

Best regards Peter

sirpete83, 2024-06-06:
ok, I got it:

C:\_PPM\Arduino_WS\hardware\espressif\esp32_v3\tools\esp32-arduino-libs\esp32/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ESP32_V3

my ESP32 is now installed in the folder esp32_v3 for testing,

and this leads to ARDUINO_ARCH_ESP32_V3 not to be ARDUINO_ARCH_ESP32
So WiFi.h is not included in your lib

sirpete83, 2024-06-06:
maybe you should use:

#include <Arduino.h>
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WebServer.h>
#include <ESP8266WiFi.h>
#endif

?

LennartHennigs, 2024-06-06:
Hey. Thanks for reaching out. 
Will try to replicate as well. Thank you.

sirpete83, 2024-06-06:
I searched through the hole lib folder. An many of the libs are selecting by this define. But if you use an other folder Name under the Hardware. I will allways fail..
It is faulty by Design

closed 2024-07-10 by LennartHennigs