LennartHennigs/ESPTelnet #73: Telnet library causes unexpected context switches

tchilton, 2024-11-23:
Whilst looking at the Telnet library on one of my projects, I have found that unexpected context switches are taking place. I am using a number of FreeRTOS tasks within my code and since using the Telnet library, unexpected switches are taking place.

I have tracked this down to EspTRelnetBase.cpp, line 32, which does a yield().
This obviously causes a context switch in FreeRTOS, hence my task which as the telnet.loop() function in it, no longer completes in one go, so effectively you have divided my timer by two - the piece up to the yield, and then next time round, the piece after the yield. 

The yield() should not be present in the library, since the library should not adversely affect other code running on the ESP32. If a yield is genuinely required, it must be in the main code, not buried within the library.

LennartHennigs, 2025-01-18:
Hey Tim,
thank you for raising the issue.
I understand you point.  
I will need to check whether removing the `yield()`does not cause other issues.
Cheers

tchilton, 2025-01-18:
Hi Lennart,

I cloned your code in as a static library in my project and then removed the the yield() line, which resolved the task issue I found. Your code seems to work fine without it in place there and with my vTaskDelay calls in place of yield(). I've now done this on 4 different modules in my system, all ESP32 based, an old NodeMCU unit, a couple of ESP32 S2 and ESP32 S3.

I suggest documenting the need for the yield() in the main code and just place this next to your telnet.loop(); function call, that way any of your tasks can run and the person implementing can decide the best place to place that task if they are doing anything more complicated. 

Great code btw, I particularly like the ANSI solution, it was cleaner than how I'd done it in my code base for my CLI's, which exist over UART, Bluetooth and Telnet.

closed 2025-01-26 by LennartHennigs

LennartHennigs, 2025-01-26:
Hey,
I removed the `yield()` and tested it briefly. It seemed to work on my end, too. 

> Great code btw I particularly like the ANSI solution; it was cleaner than how I'd done it in my code base for my CLI's, which exist over UART, Bluetooth, and Telnet.

Thanks, I took quite some inspiration from Bruce Robertson's ANSITerm. 
The link is in the header file.